stm32duino/STM32Examples #9: Too much time spent on .loop method

MacsimBui, 2018-12-11:
With the MQTT PubSubClient example, the B-L475E-IOT1A spends 10 seconds within the client.loop() method and processes messages with a very long delay : it cannot process messages fast enough.
Is that an issue or is there any parameter to make it process incoming messages faster ?

fpistm, 2018-12-11:
Hi,

IIRC, pubsubclient use available() API to know if there is some data available. unfortunatelly, the WiFi library alaways return 1 so pubsubclient always try to compute data which I guess drive to several lost of time.
I solution could be to adapt pubsubclient to avoid this.

fpistm, 2019-03-27:
Will be covered with https://github.com/stm32duino/WiFi-ISM43362-M3G-L44/issues/10

closed 2019-03-27 by fpistm