stm32duino/STM32Examples #11: Stm32F103C8T6

nikjayswal, 2019-05-14:
Hi, i am building an fast TFT Oscilloscope i used analogRead but it is slow i want to use interrupt as a analogRead but i have no idea how to do this.

nikjayswal, 2019-05-15:
My code : HardwareTimer timer(3);

#define USE_NIKJAYSWAL_BLUEPILL

#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h>    // Core graphics library
#include <MCUFRIEND_kbv.h>
#include "TouchScreen_kbv.h"

#include <Fonts/FreeSans9pt7b.h>
#include <Fonts/FreeSans12pt7b.h>
#include <Fonts/FreeSerif12pt7b.h>

#include <FreeDefaultFonts.h>

#define LCD_CNTRL GPIOA
#define LCD_DATA  GPIOB

#define LCD_CS PA0 // Chip Select goes to Analog 3
#define LCD_CD PA1 // Command/Data goes to Analog 2
#define LCD_WR PA2 // LCD Write goes to Analog 1
#define LCD_RD PA3 // LCD Read goes to Analog 0
#define LCD_RESET PA8 // Can alternately just connect to Arduino's reset pin

#define LCD_D0 PB6
#define LCD_D1 PB7
#define LCD_D2 PB8
#define LCD_D3 PB9
#define LCD_D4 PB10
#define LCD_D5 PB11
#define LCD_D6 PB14
#define LCD_D7 PB15

#define BLACK 0x0000
#define NAVY 0x000F
#define DARKGREEN 0x03E0
#define DARKCYAN 0x03EF
#define MAROON 0x7800
#define PURPLE 0x780F
#define OLIVE 0x7BE0
#define GREY  0x8410
#define LIGHTGREY 0xC618
#define DARKGREY 0x7BEF
#define BLUE 0x001F
#define GREEN 0x07E0
#define CYAN 0x07FF
#define RED 0xF800
#define MAGENTA 0xF81F
#define YELLOW 0xFFE0
#define WHITE 0xFFFF
#define ORANGE 0xFD20
#define GREENYELLOW 0xAFE5
#define PINK 0xF81F

MCUFRIEND_kbv tft;

volatile int IRQcount;
int pin_irq = PA6; //IRQ that matches to pin 2
volatile int buffer1[242];
volatile int buff_val = 0;
volatile int pin;

void setup() {
  // put your setup code here, to run once:
  Serial.begin (115200);
  noInterrupts;
  timer.pause();
    timer.setCount(0);
    timer.setOverflow(30000);
    timer.setCompare(TIMER_CH2, 65535);   // somewhere in the middle
  pinMode(pin_irq, INPUT_PULLDOWN);
  attachInterrupt(pin_irq, IRQcounter, FALLING);
  timer.refresh();
    timer.resume();
    interrupts;
}

void IRQcounter() {

  IRQcount++;
  pin = analogRead(pin_irq);
  pin = map(pin, 0, 4095, 240, 0);
  buffer1[buff_val] = pin;
  buff_val = buff_val + 1;
  for (int i=0; i<240; i++){
    buffer1[i];
     
  }
 
  if(buff_val > 241)  {
    buff_val = 0;
  }
}

void loop() {

  IRQcount = 0;
  //Serial.print(F("Counted = "));
  for(int i = 0; i<240 ; i++) {
  Serial.println(buffer1[i]);
  delayMicroseconds(10000);
  }
  noInterrupts;
}

my code not showing Arduino Serial plotter Osciloscope properly i am using 50Hz shinewave frequency but when i am using simple osciloscope without interrupe it showing properly but its is slow also osciloscope not zoom to determine frequency waveforms , so i am searching on google for interrupt based osciloscope but not find anything either Ping-O-Scope but this scop is not easy also not supported to my Mcufriend Library does have any idea i am using 2.4 tft display - tft ID : 0x4747

fpistm, 2019-05-15:
Hi @nikjayswal 
This library relies on this core: https://github.com/stm32duino/Arduino_Core_STM32
I guess you use this one: https://github.com/rogerclarkmelbourne/Arduino_STM32
You should open anissue on this GitHub.

nikjayswal, 2019-05-15:
Thank you fpistm may be given link is helpful.

Thanks again

closed 2019-05-15 by nikjayswal

fpistm, 2019-05-15:
welcome