stm32duino/STM32Examples #22: Timebase_callback.ino example not working

TadyTheFish, 2020-02-28:
Hello 
I would just like to point out that this example is not working for me with stm32f103 bluepill.
To make it work you need to put `MyTim->setMode(2, TIMER_OUTPUT_COMPARE);` just before
`MyTim->setOverflow()`
Maybe some one could correct this?

Thank you

fpistm, 2020-02-28:
Hi @TadyTheFish 
this is not an issue.
If you use the latest release instead of the master archive it is OK.
With next core version 1.9.0 the `setMode` is no more needed, see #18

closed 2020-02-28 by fpistm

joej970, 2020-03-08:
@TadyTheFish 
Thanks for pointing that out. It was driving me crazy for past 2 hours. 
@fpistm 
Can you tell me where I find the latest release so the example works without extra setMode() line? [https://github.com/stm32duino/Arduino_Core_STM32/releases](https://github.com/stm32duino/Arduino_Core_STM32/releases)  last release is 1.8.0 which is the version I have updated to through Boards manager but it does not work without extra setMode() line.

TadyTheFish, 2020-03-08:
Select the core that you are using and go to Examples and look for Timebase_callback. The examole that is inside your IDE is correct.. This one on the webpage is missing the setMode.
For now this one on the webpage does not work but it will with the 1.9.0 release. So add setMode() or wait for the 1.9.0 release

joej970, 2020-03-08:
Thanks, but I cannot find the examples in IDE. Under Examples for generic STM32F1 series I only have EEPROM, IWatchdog, Servo, SoftwareSerial, SPI, Wire. Neither can I locate it in packages/hardware/stm32/1.8.0 directory where is location of the STM32 Core files. 

However I am more insterested in getting the latest release as advised by @fpistm.

fpistm, 2020-03-08:
The 1.9.0 is not yet officially released. Same for STM32Examples library.
If you install this library thanks the library manager then you will have the example inline with the last core version. When I will release the 1.9.0 then I will release also a new version of this library.

TadyTheFish, 2020-03-08:
Do you know when you will release 1.9.0? 6 months, year?

fpistm, 2020-03-08:
weeks