stm32duino/STM32Examples #24: Timebase_callback.ino does not work on Bluepill STM32F103

HamidSaffari, 2020-04-01:
Hi there.
Your Hardwre Timer example callback do not execute the callback function at all.
https://github.com/stm32duino/STM32Examples/blob/master/examples/Peripherals/HardwareTimer/Timebase_callback/Timebase_callback.ino

fpistm, 2020-04-01:
See #22 #20 #19 ...

HamidSaffari, 2020-04-01:
Thanks man, we are waiting to release the 1.9.0 version and make all us happy specially with the all new device supports which almost cover every available boards and rock the Arduino world. and make us easy to work with fast MCUs than working with lots of headers and src files in professional IDE projects.

closed 2020-04-01 by fpistm

HamidSaffari, 2020-04-01:
I put your today master repository but got this error:

no matching function for call to 'HardwareTimer::attachInterrupt(void (&)(HardwareTimer*))'

fpistm, 2020-04-01:
As explain in other issue, use the official release not the master.
The release is aligned with Core release. The master with core master.
There is a API change for callback. See the PR #23

HamidSaffari, 2020-04-01:
Yes I got it just change the  
  void Update_IT_callback(HardwareTimer*)   to
  void Update_IT_callback()      or      void Update_IT_callback(void)
and that change makes it look like other Arduino attach interrupts.