stm32duino/STM32Examples #25: Timebase callback example

monoplex, 2020-05-01:
Hi,
   When trying the 'Timebase callback' example on a STM32F030F4P6 demo board with STM32duino, I'm sorry I just don't get the significance of the:

#if !defined(STM32_CORE_VERSION) || (STM32_CORE_VERSION  < 0x01090000)
#error "Due to API change, this sketch is compatible with STM32_CORE_VERSION  >= 0x01090000"
#endif

This throws up a compilation error but I am unable discover what the STM32_CORE_VERSION number refers to, what it might be for me or how to fix the cause of the error (I have searched).

Where might I read about "Due to API change"?

I suppose the bottom line is, where can learn how to use the hardwaretimers library correctly?

   Thanks.

fpistm, 2020-05-01:
This means you don't use the correct core version to use this example.
https://github.com/stm32duino/Arduino_Core_STM32
You probably use the master of this library but it is not yet release (tag) and will be compatible with the next release of the core 1.9.0.

So if you use the core 1.8.0 then use the latest release of this library.

monoplex, 2020-05-02:
Right! I've got it straight now. Thanks for your words.

closed 2020-05-02 by monoplex