stm32duino/STM32Examples #37: HardwareTimer - is there an Interrupt, when it overflows?

SunboX, 2021-02-05:
Hi,

I have an external event (clock) source. I want to count the pulses. Is there an interrupt, when the counter overflowes?
This is, what I tried so far:

```c++
#include <Arduino.h>

#define pin PA8

uint32_t channel;
static volatile HardwareTimer *MyTim;
static volatile uint32_t timestamp;

void InputCapture_IT_callback(void)
{
  timestamp = micros();
}

void Rollover_IT_callback(void)
{
  digitalWriteFast(PB_4, HIGH);
  Serial.println("rollover");
  MyTim->setCount(0);
  MyTim->pause();
}

void setup()
{
  pinMode(PB4, OUTPUT);

  pinMode(pin, INPUT_PULLUP);
  digitalWrite(pin, HIGH);

  // Automatically retrieve TIM instance and channel associated to pin
  // This is used to be compatible with all STM32 series automatically.
  TIM_TypeDef *Instance = (TIM_TypeDef *)pinmap_peripheral(digitalPinToPinName(pin), PinMap_PWM);
  channel = STM_PIN_CHANNEL(pinmap_function(digitalPinToPinName(pin), PinMap_PWM));

  MyTim = new HardwareTimer(Instance);
  MyTim->pause();
  MyTim->setMode(channel, TIMER_INPUT_CAPTURE_FALLING, pin);
  MyTim->setPrescaleFactor(1);

  MyTim->setOverflow(288, TICK_FORMAT); // count up to 288 bits

  MyTim->attachInterrupt(channel, InputCapture_IT_callback);
  MyTim->attachInterrupt(Rollover_IT_callback);
  MyTim->setCount(0);
  MyTim->refresh();
  MyTim->resume();
}

void loop()
{
  uint32_t timeDiff = micros() - timestamp;

  if (timeDiff > 300) // 300us -> reset
  {
    MyTim->setCount(0);
    MyTim->refresh();
    MyTim->resume();
    
    digitalWriteFast(PB_4, LOW);
  }
}
```

SunboX, 2021-02-05:
The `rollover` should be this event, or? But it happens way to often.

if this is my signal on PA8
```
  __    __    __    __    __    __    __                   __    __    __    __    __    __    __
_|  |__|  |__|  |__|  |__|  |__|  |__|  |_________________|  |__|  |__|  |__|  |__|  |__|  |__|  |_______________
 1.25us      1.25us      1.25us            300us delay    1.25us      1.25us
```

I want to achive this on PB_4
```
                    _____________________________________                    ____________________________________
___________________|                                     |__________________|                   
                   288 bit interrupt                     reset after 300us timeout
```

SunboX, 2021-02-05:
The signal is generated from anothe uC using the Adafruit Neopixel library.

SunboX, 2021-02-05:
I did a small test skatch for the signal providing uC, just to slow down everything. This is the test code:

```c++
#define LED_PIN 14
#define NUMBER_OF_LEDS 24

int i = 0;

void setup() {
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);

  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);
}

void loop() {
  i++;
  
  digitalWrite(LED_PIN, HIGH);
  digitalWrite(LED_BUILTIN, HIGH);
  delay(8);
  digitalWrite(LED_PIN, LOW);
  digitalWrite(LED_BUILTIN, LOW);
  delay(4);

  if ( i == NUMBER_OF_LEDS * 24) {
    delay(500);
    i = 0;
  }
}
```

If I change now the "tick" interrupt like this and remove the whole loop-code
```c++
void InputCapture_IT_callback(void)
{
  uint32_t timeDiff = micros() - timestamp;
  Serial.print("tick ");
  Serial.println(timeDiff);
  timestamp = micros();
}
```

the console output is as expected:
```
tick 12009
tick 12012
tick 12011
tick 12011
tick 12011
tick 12012
tick 12012
tick 12012
tick 12473
tick 12041
tick 12014
tick 12012
tick 12012
tick 12009
tick 12011
tick 12011
tick 12012
tick 12012
tick 12012
tick 12011
tick 12011
tick 12011
tick 12011
tick 12011
tick 12011
tick 12011
tick 12011
tick 12011
```

But if I comment out the "rollover" (overflow?) interrupt like this
```c++
void Rollover_IT_callback(void)
{
  uint32_t timeDiff = micros() - timestamp;
  Serial.print("rollover ");
  Serial.println(timeDiff);
  timestamp = micros();
}
```

I'm getting this console output:
```
rollover 5
rollover 7
rollover 6
rollover 5
rollover 5
rollover 5
rollover 5
```

In my opinion, this shouldn't be right. What I'm doing wrong? I had expected something like 288 * 12ms = 3456ms (3456000us)

ABOSTM, 2021-02-05:
Hi @SunboX,
First, it would be better for next time to raise such question on the forum:
https://www.stm32duino.com/

Then I think there is a misunderstanding in the use of Input Capture:
In this mode, counter is clocked by internal source clock, and then falling edge on the input pin (or rising depending on the chosen configuration), will capture the value of the counter into Capture register. So this feature is designed to measure time between input edges (or implement timeout).
But as I understand you try to use it to count the number of pulse (number of bit). It is not possible with input capture implemented in Arduino.
When you set 288 ticks, it means: 
288* Timer internal clock. I guess your use default L412KB default variant which means 80MHZ as system clock and in this case also 80MHz as timer internal clock.
288* 1/80000000 = 0.0000036s = 3,6micro second

Several option:
* Use simply an interrupt on pin (without Hardware timer). And increment a counter in corresponding interrupt. This is quite simple, but the drawback is that you get a software interrupt at each falling edge and so this increase cpu load.
* If you absolutely need hardware processing to free CPU, it is possible using STM32 timer, but t is not implement in Arduino, so you need to use STM32Cube HAL to do so (STM32 Cube HAL can be used in Arduino sketch directly, but in this case use of HardwareTimer is prohibited or should concern a separate timer ). It is much more efficient, but drawback: it is more complex.

SunboX, 2021-02-05:
Hey, thanks for your help. Yes you are right, I thought it's an issue. But it is some less knowledge of the STM32 Arduino impl. on my side. Sorry for that. Thanks for your explanation!
Now it's more clear to.me, why it works like this.

Am I allowed to ask one more question? 😅

Do you have some sample code how the STM32 Cube HAL can be used inside the Arduino sketch directly to count the bits? It's bit difficult to find a sample in Google, seems a bit special.

SunboX, 2021-02-07:
btw., thats my current code. But for some reason it does not work. I had to instantiate the HardwareTimer because it defines all the needed interrupts. :( Or do you have any suggest how to avoid using it? I would still like to use the Arduino Framework because of other libs and I don't want to block the main program. So I really need the hardware timer for counting.

```c++
#include <Arduino.h>

#define pin PA0

static volatile uint32_t timestamp;

void tickCallback(void)
{
  uint32_t timeDiff = micros() - timestamp;
  Serial.print("tick ");
  Serial.println(timeDiff);
  timestamp = micros();
}

void periodElapsedCallback(void)
{
  Serial.println("overflow reached");
}

void setup()
{
  Serial.begin(115200);

  pinMode(pin, INPUT_PULLUP);

  // Automatically retrieve TIM instance and channel associated to pin
  // This is used to be compatible with all STM32 series automatically.
  TIM_TypeDef *Instance = (TIM_TypeDef *)pinmap_peripheral(digitalPinToPinName(pin), PinMap_PWM);
  uint32_t channel = STM_PIN_CHANNEL(pinmap_function(digitalPinToPinName(pin), PinMap_PWM));

  HardwareTimer *MyTim = new HardwareTimer(Instance);
  TIM_HandleTypeDef *htim2 = MyTim->getHandle();

  MyTim->attachInterrupt(periodElapsedCallback);
  MyTim->attachInterrupt(channel, tickCallback);

  htim2->Init.Prescaler = 0;
  htim2->Init.CounterMode = TIM_COUNTERMODE_UP;
  htim2->Init.Period = 288;
  htim2->Init.ClockDivision = TIM_CLOCKDIVISION_DIV1;
  htim2->Init.AutoReloadPreload = TIM_AUTORELOAD_PRELOAD_ENABLE;
  if (HAL_TIM_Base_Init(htim2) != HAL_OK)
  {
    Serial.println("HAL_TIM_Base_Init error");
  }

  TIM_ClockConfigTypeDef sClockSourceConfig = {0};

  sClockSourceConfig.ClockSource = TIM_CLOCKSOURCE_ETRMODE2;
  sClockSourceConfig.ClockPolarity = TIM_CLOCKPOLARITY_NONINVERTED;
  sClockSourceConfig.ClockPrescaler = TIM_CLOCKPRESCALER_DIV1;
  sClockSourceConfig.ClockFilter = 0;
  if (HAL_TIM_ConfigClockSource(htim2, &sClockSourceConfig) != HAL_OK)
  {
    Serial.println("HAL_TIM_ConfigClockSource error");
  }

  TIM_MasterConfigTypeDef sMasterConfig = {0};
  sMasterConfig.MasterOutputTrigger = TIM_TRGO_RESET;
  sMasterConfig.MasterSlaveMode = TIM_MASTERSLAVEMODE_DISABLE;
  if (HAL_TIMEx_MasterConfigSynchronization(htim2, &sMasterConfig) != HAL_OK)
  {
    Serial.println("HAL_TIMEx_MasterConfigSynchronization error");
  }

  HAL_TIM_Base_Start_IT(htim2);
}

void loop()
{
  Serial.println(TIM2->CNT);
}
```

ABOSTM, 2021-02-08:
Hi @SunboX,
I am glad to see that you succeeded to add some HAL code in Arduino sketch, for sure there is no dedicated 'sample code how the STM32 Cube HAL can be used inside the Arduino sketch directly to count the bits'.

Please note that if you go that way (HAL usage), it supposes some knowledge of STM32 Timer. So, if not already done, I strongly suggest you to read TIM chapter of your Reference Manual.

You figure out that you need to use ETR mode 2 for counting on external clock. Great.
But be careful: all Timer instances doesn't have the same capabilities. And specially all instances doesn't have ETR capability.
So make sure the instance you use support it.
Also please note that ETR is a dedicated pin (different from TIMx_CHy) so you need to configure corresponding gpio pin manually : it means do it with STM32 HAL: it could not be done by Arduino, as Arduino doesn't support ETR. 
And of course, connect this pin to your external clock source. 
I think this is you major issue.

> I had to instantiate the HardwareTimer because it defines all the needed interrupts.

Whatever if it works for you with HardwareTimer instanciation (and only instanciation) that fine.

fpistm, 2021-02-12:
I consider this as answered and close this issue as it is not related to a bug.
Support or question have to be raised on the forum.

closed 2021-02-12 by fpistm