stm32duino/STM32Examples #39: Example issue

alejoseb, 2021-02-11:
Hi is the following example working?

https://github.com/stm32duino/STM32Examples/blob/master/examples/Boards/STM32L475VG-DISCOVERY-IOT/BTLE_sensors_TimeOfFlight_demo/BTLE_sensors_TimeOfFlight_demo.ino

I notice the constructor for VL53L0X is implemented only with 2 parameters but 3 are passed in the example.
Is there any updated version to test the VL53L0X?

Also the example provided in the Cube package for B-L475E-IOT01A is not working.

Thanks.

fpistm, 2021-02-12:
Hi @alejoseb 
you are right with the new version of the library  on parameters has been removed by @cparata in https://github.com/stm32duino/VL53L0X/pull/13
Example is compatible with  version < 2.0.0.

cparata, 2021-02-12:
Hi @fpistm ,
if we need to update the example, I prefer to adapt it to the new APIs and that's it. In my opinion, it does not make sense to keep it compatible with old APIs because we improve the library to avoid it and then we make the example more complex without any reason. Anyway the fix is very simple. You just need to remove the 3rd parameter in the constructor (that was not used in the class, so it was useless) and add the call to "begin" API in the setup function.
Best Regards,
Carlo

cparata, 2021-02-12:
I generated the PR to update the example to the new APIs. Unfortunately, I do not have the control of all examples that use the sensor APIs. So, I suggest to the owners of the several examples based on the sensor APIs to update the sketches to the new APIs. It should not be a big job.
Best Regards,
Carlo

fpistm, 2021-02-12:
OK. Thanks.

closed 2021-02-12 by fpistm