stm32duino/STM32Examples #42: setPWM(...) causes STM32F103C8 code to hang

kevintedder, 2021-03-14:
I'm using the hardwaretimer library to control a PWM signal to driver a stepper motor.  My main loop code continuously calculates the PWM frequency from the desired RPM for the motor.  However, using the setPWM(...) function within the main loop causes my code to crash and hangs the CPU.
After many hours of head scratching I have found the solution.  The setPWM(...) function requires a Pause() at the beginning, before changing the timer settings. The resume(), at the end, then allows it to continue.

Can you please include this minor change in the next release.
Thx.

FYI - it should look like this

void HardwareTimer::setPWM(uint32_t channel, PinName pin, uint32_t frequency, uint32_t dutycycle, callback_function_t PeriodCallback, callback_function_t CompareCallback)
{
//#######################
  pause();
//#######################
  setMode(channel, TIMER_OUTPUT_COMPARE_PWM1, pin);
  setOverflow(frequency, HERTZ_FORMAT);
  setCaptureCompare(channel, dutycycle, PERCENT_COMPARE_FORMAT);
  if (PeriodCallback) {
    attachInterrupt(PeriodCallback);
  }
  if (CompareCallback) {
    attachInterrupt(channel, CompareCallback);
  }
  resume();
}

ABOSTM, 2021-03-15:
Hi @kevintedder,
The real issue is that you use `setPWM()` function to update duty-cycle/frequency of a running PWM.
It is not designed for that, but only to simply PWM configuration in one single function starting for scratch.
What you should do is:
* use `setPWM()` only for the 1st time
* use `setCaptureCompare()` or `setOverflow()` to update dutycycle or frequency

Thus you don't need to pause the motor to change its rotation parameters.

kevintedder, 2021-03-15:
@fpistm noted. Many thanks. I'll make the necessary code change.
Can I suggest that the All in one PWM example be updated to show how to do this to help others in the future. This was what I based my code on.
😀

ABOSTM, 2021-03-15:
> @fpistm noted. Many thanks. I'll make the necessary code change.
> Can I suggest that the All in one PWM example be updated to show how to do this to help others in the future. This was what I based my code on.

I prefer not, because I want to keep this "All in one PWM example" as simple as possible with a single function call `setPWM()`

By the way please note that wiki exist for HardwareTimer library: 
https://github.com/stm32duino/wiki/wiki/HardwareTimer-library

closed 2021-03-15 by fpistm