stm32duino/STM32Examples #62: All-in-one setPWM problem

Omidttl, 2025-01-31:
All-in-one setPWM example doesn't work on bluepill

fpistm, 2025-01-31:
Could you be more precise on the problem you face?
Which pinyou used? Options selected?....

Omidttl, 2025-01-31:
this is my code.it compiles well and uploads correctly on the blupill, but nothing shows on my oscilloscope 

```
/*
  All-in-one setPWM
  This example shows how to configure a PWM with HardwareTimer in one single function call.
  PWM is generated on `LED_BUILTIN` if available.
  No interruption callback used: PWM is generated by hardware.
  Once configured, there is no CPU load.
*/

/*
  Note: Please verify that 'pin' used for PWM has HardwareTimer capability for your board
  This is specially true for F1 serie (BluePill, ...)
*/

// #if defined(LED_BUILTIN)
// #define pin  LED_BUILTIN
// #else
// #define pin  D2
// #endif
// #define pin  PC13
#define pin  PA0

void setup()
{
  // no need to configure pin, it will be done by HardwareTimer configuration
  pinMode(pin, OUTPUT);

  // Automatically retrieve TIM instance and channel associated to pin
  // This is used to be compatible with all STM32 series automatically.
  TIM_TypeDef *Instance = (TIM_TypeDef *)pinmap_peripheral(digitalPinToPinName(pin), PinMap_PWM);
  uint32_t channel = STM_PIN_CHANNEL(pinmap_function(digitalPinToPinName(pin), PinMap_PWM));

  // Instantiate HardwareTimer object. Thanks to 'new' instantiation, HardwareTimer is not destructed when setup() function is finished.
  HardwareTimer *MyTim = new HardwareTimer(Instance);

  // Configure and start PWM
  // MyTim->setPWM(channel, pin, 10, 50, NULL, NULL); // No callback required, we can simplify the function call
  MyTim->setPWM(channel, pin, 5, 50); // 5 Hertz, 10% dutycycle
}

void loop()
{

  /* Nothing to do all is done by hardware. Even no interrupt required. */
}
```

fpistm, 2025-01-31:
And which board you select?

fpistm, 2025-02-03:
Hi @Omidttl 
I've tested your sketch and works as expected:

[image]
Are you sure you have a legacy STM32F103C8 on your BluePill.

closed 2025-02-03 by fpistm